Ircama/ELM327-emulator #4: Does it contain bluetooth communication?

saramonteiro, 2019-11-09:
Hi, I realized it emulates a ELM327 adapter but using a USB communication. So, does it contain a version that is able to communicate to the application using bluettooth, because many adapters talk through bluetooth.
thanks in advance!

Ircama, 2021-01-09:
You should connect your application to the tty/COM port provided by ELM327-emulator, which is agnostic of the physical interface (USB or Bluetooth) and just exploits a virtual tty/COM port.

Ircama, 2021-03-14:
I think this remark was related to the *obd_dictionary* tool, which is now provided with all the python-OBD interface settings (see `-B`, `-T`, `-C`, `-F`, `-P` command-line options). It also has a dry-run flag (`-r`), which is very useful to test the OBDII connection.
The [documentation](https://github.com/Ircama/ELM327-emulator#elm327-emulator-batch-mode) has also been improved, including some notes on Bluetooth.

closed 2021-03-14 by Ircama

Ircama, 2021-03-28:
Added Bluetooth documentation

commit dd5dddbd (2021-03-14): Version 1.1.1

Dry-run option in obd_dictionary.

New pids for 'car' scenario.

Updated documentation.

resolves #4