Ircama/ELM327-emulator #5: emulator starts at /dev/ttys000 on Mac OS Mojave

oleksandr-k, 2020-02-11:
I'm using emulators as described and it starts at /dev/ttys000. I'm using connector for serial ports interaction and it can't find the device. Can I make somehow to start emulator at /dev/tty.XXX?

Ircama, 2020-02-12:
Can you test this with python3?
```python
import pty
import os
master_fd, slave_fd = pty.openpty()
print("Running on", os.ttyname(slave_fd))
```

oleksandr-k, 2020-02-12:
Hello @Ircama , thanks a lot for your attention, I'm sorry for bothering you, currently I avoid the problem by using another 3rd party for accessing to serial port on Mac.
I used ORSSerialPort library, it's good but it detects ports only on tty.xxx so I switched to SwiftSerial and it handles ttysXXX.

Thanks!

closed 2020-02-12 by oleksandr-k