Ircama/ELM327-emulator #7: trouble with emulator.answer

crispyjones, 2021-02-16:
I have this up and running on a linux machine and I am getting valid values on the arduino it's connected to. It's a barebones sketch printing out RPM and coolant temp. I get non-zero values for coolant in default scenario and both rpm and coolant temp in the car scenario. When I copy paste your example RPM  line `emulator.answer['RPM'] = '\0 ECU_R_ADDR_E + " 04 41 0C %.4X" % int(4 * 500) \0\r'` I get the following reply `Error executing command: name "emulator" is not defined` with no change to the values I see on the arduino terminal. I tried various other emulator.answer examples all with the same result. Other emulator commands I discovered with tab complete work like emulator.get_pty, just not emulator.answer. Tried installs from git and pip with no difference.
I'm a car guy with very little programming knowledge, I expect I'm doing something obviously wrong but I can't figure it out.

crispyjones, 2021-02-18:
I went back to [this commit](https://github.com/Ircama/ELM327-emulator/tree/196cd7cc4d185379904224ebb31668881cabfcfd). After correcting my mistake with the RPM sample to use the hex address of the ECU from obd_message.py like this `emulator.answer['RPM'] = '\0 "7E8" + " 04 41 0C %.4X" % int(4 * 500) \0\r'` it worked. Arduino now displays 500, or whatever I put in there, for the RPM. To satisfy my curiosity I reinstalled the latest version and copy/pasted that exact command and it still returns `Error executing command: name "emulator" is not defined`.

Ircama, 2021-02-18:
@crispyjones, thanks for notifying this issue in the new version: it needs to be fixed.

closed 2021-02-19 by Ircama

Ircama, 2021-02-19:
[ELM327-emulator 1.0.2.post11](https://pypi.org/project/ELM327-emulator/) published, which should fix the bug.

commit 5d7d4de3 (2021-02-19): Fix execution of commands including "emulator" variable
Resolves #7