Ircama/ELM327-emulator #9: UDS support

mickeyl, 2021-03-07:
First off, thanks for this great project. I'm working on various scan tools myself and did work on some ad-hoc emulators. It might make sense to join forces on one common project. Right now I'm after UDS support, i.e. answers to service IDs >= 0x10.

I realize that would lean the project more towards ECU simulation, for which some other interesting alternatives exist, but figure I'd just ask what you are thinking about that.

Ircama, 2021-03-07:
@mickeyl thanks.

I'm not experienced in UDS support, can you provide some examples of requests/responses?

In any case, this project is open to contributions. If you have ideas to improve *ELM327-emulator*, I invite you to create a topic branch and develop there the additions.

mickeyl, 2021-03-17:
UDS communicates pretty much in the same way as OBD2, as a matter of fact, OBD2 is actually a subspec of UDS that covers the PIDs 01-09, whereas UDS deals with PIDs onwards 0x10. Here's a quick commented dump of the start of a reprogramming session:

```
// Extended Diagnostics Session
7E5 02 10 03
7ED 06 50 03 00 14 00 C8
// Hardware Reset
7E5 02 11 01
7ED 03 7F 11 78
7ED 02 51 01
// Programming Session
7E5 02 10 02
7ED 06 50 02 00 14 00 C8
// Request Seed
7E5 02 27 11
7ED 03 7F 27 78
7ED 06 67 11 A6 41 B5 E9
// Send Key
7E5 06 27 12 B1 51 D5 8F
7ED 02 67 12
// Write Fingerprint (2E: F15A => 0004140606FFFFFFFF)
7E5 10 0C 2E F1 5A 00 04 14
7ED 30 20 00
7E5 21 06 06 FF FF FF FF 00
7ED 03 6E F1 5A
// Routine Control: Start (01), Delete Area (FF 00), Bootloader (01 00)
7E5 06 31 01 FF 00 01 00
7ED 03 7F 31 78
7ED 03 7F 31 78
7ED 03 7F 31 78
7ED 05 71 01 FF 00 00
```

… and so on.

Ircama, 2021-03-18:
@mickeyl thanks, interesting. At first glance, it looks pretty easy to extend the dictionary to support this. I assume you only added the RX commands in the example, I can't find the TX part. Ideally, it would be useful through some application to get the whole traces of a complete set of queries to real ECUs for all possible commands; please suggest a tool if exists, consider also that, if I have to run it myself, I can only use a real engine and I suspect some queries might be invasive (I would like to avoid reprogramming an ECU which modifies the performance, or damages the engine or invalid the warranty). Otherwise, if you are able to get a long enough listing of TX/RX flow, we could simply start from this, trying to convert it into a dictionary. And we need a trace of a whole session since the beginning, also including the preliminary AT commands. (Notice that the dictionary format is being changed in a future version of ELM327-emulator, which will include ATH0/ATH1 support.)

mickeyl, 2021-03-18:
There are no special AT commands necessary. As I've said, UDS is a super spec of OBD2, hence the adapter-level setup is completely the same – apart from setting TX and RX headers (ATSH + ATCRA).

The part I have quoted indeed lists both TX and RX. It's from a real conversation between a scan tool (0x7E5) and a power train ECU (0x7ED). I need to check in how far I can give you more of the logs, since they contain potentially confidential information.

Ircama, 2021-04-10:
@mickeyl thanks for your clarifications!

[Version 3](https://github.com/Ircama/ELM327-emulator/tree/v3.0.0) (a major upgrade, still in beta) also allows the UDS support. It takes advantage of a plugin architecture to separately manage [tasks](https://github.com/Ircama/ELM327-emulator/tree/v3.0.0#tasks). In principle, UDS communication should be implemented through the development of dedicated plugins ([some of them are available](https://github.com/Ircama/ELM327-emulator/tree/v3.0.0/elm/plugins) as basic examples). You can check the related branch, which should be able to run your examples.

```
python3 -m pip install git+https://github.com/ircama/ELM327-emulator.git@v3.0.0
```

I still do not know a way to configure OBD-II in order to directly send the requests like in your example (where all of them already include ECU ID and size instead of doing *ATSH* to change the header):

```
7E5 02 10 03
```

(this instead of `ATSH7E5` and then `10 03`, or `02 10 03` if using `ATCAF0`).

Is it `ATFCSM1`?

mickeyl, 2021-04-12:
@Ircama I guess I wasn't clear enough … the log I have shown was not a serial log, but a can-log. Can-Logs just print the CAN id on the left. When communicating with the adapter, I'm really just sending the usual pair of: `ATSH` and `ATCRA`, and then the payload.

When it comes to supporting UDS in the emulator, there are (among others) two important things to consider.

1. UDS is stateful. You enter privileged diagnostic sessions (optionally security guarded by a challenge/response exchange), then communicate. The UDS server (ECU) is falling back to the default – unprivileged – diagnostic session after a communication timeout.

2.) ISO-TP / Flow Control. I guess this is the most delicate part for the adapter (simulator), since in contrast to OBD2, UDS is about sending up to 4095 payload bytes per command, fragmented via ISO-TP into 586 CAN frames and – by default – every 32 blocks the device is expecting a flow control answer from the adapter (simulator).

Since ISO-TP comes with strong timing guarantees, it is not feasible to perform UDS on an adapter that comes without built-in ISO-TP encoding. This rules out pretty much every "standard" ELM327 adapter, since they usually do not have this.

STN11xx, STN22xx, etc., based adapters (which pretty much emulate most of the ELM327 command set plus some extra commands) have that – so to fully implement UDS in the emulator, we would have to offer this as well, i.e. pretend that we can deal with commands up to 4095 bytes.

"From the outside", an adapter that has builtin support for ISO-TP encoding just takes frames longer than 8 bytes and formats them before sending consecutive frames to the device.

Ircama, 2021-04-12:
> I'm really just sending the usual pair of: `ATSH` and `ATCRA`, and then the payload.

`ATSH` is already supported in version 2. `ATCRA` works in version 3.

----------

> 1. UDS is stateful.

This is the reason why version 3 introduces [tasks](https://github.com/Ircama/ELM327-emulator/tree/v3.0.0#tasks), which are stateful objects with proper namespace, that can remain active for the whole duration of a specific diagnostic session, can run concurrently on different EQU IDs and are dynamically released, including resources. When a task is started on an EQU ID upon an application request (a CAN request that is mapped to a task in the dictionary), the application communication to that EQU ID is redirected to the task and managed by it. You can define a task that directly implements a session workflow, or that interfaces a thread or an external backend process.

Until version 2, *ELM327-emulator* was just based on a stateless request/response dictionary, allowing execution of single-line code within the same namespace of the emulator.

----------

> 2. ISO-TP / Flow Control.

For basic communication, there is already trivial support for this, including periodic flow control answers. You can find some mention in the [new documentation](https://github.com/Ircama/ELM327-emulator/tree/v3.0.0#helper-functions).

----------

> Since ISO-TP comes with strong timing guarantees, it is not feasible to perform UDS on an adapter that comes without built-in ISO-TP encoding.

For more complex communication, a task might possibly allow integration of specific ISO-TP libraries or backend services.

----------

> STN11xx, STN22xx, etc., based adapters (which pretty much emulate most of the ELM327 command set plus some extra commands) have that

A tiny subset of 'ST' commands is included in version 3.

Ircama, 2021-06-12:
Resolved by https://github.com/Ircama/ELM327-emulator/pull/11

closed 2021-06-12 by Ircama