Ircama/ELM327-emulator #10: (Feature) Serve commands via network

mickeyl, 2021-03-18:
Many ELM327-based OBD2 adapters work via WiFi. It would be cool to have the ELM327-emulator listen via a customizable port. That way it could serve as a testbed for OBD2 apps running on mobile clients which usually can't connect via serial port.

Ircama, 2021-03-18:
This is doable in principle but takes some coding and testing work (at the moment this application only supports traditional pseudo-tty ports and we need to bridge an inet socket to the pty code to do that). Can you make an example of a mobile app and related TCP/IP port?

mickeyl, 2021-03-19:
All apps that support WiFi-based ELM327 adapters connect via `192.168.0.10:35000` – there are tons in the respective iOS and Android app stores.

Ircama, 2021-03-21:
@mickeyl , yes, obviously, thanks for making me remind of this method. I implemented a preliminary version of the program supporting TCP/IP via `-n` option.

commit a9a32307 (2021-03-21): TCP/IP Support to serve commands via network

Closes #10

Reference #4
closed 2021-03-21 by Ircama