Ircama/ELM327-emulator #13: CRITICAL - Error while processing '': string index out of range

noloader, 2021-07-10:
Hi Everyone,

Thanks for this emulator. It looks like it will be very useful.

I'm working on Ubuntu 20.04, x86_64, fully patched. Python 3.8.10 is installed, and dependencies are installed. I cloned `ELM327-emulator`, and ran it using `python3 -m elm`. The emulator was waiting/listening on `/dev/pts/3`.

I then connected my client to the emulator via `/dev/pts/3`. I issued a `ATZ` from my client to reset the controller. The command hung in my client.

At the emulator there was a lot of activity:

```
$ python3 -m elm
2021-07-09 21:25:42,876 - root - INFO - 

ELM327 OBD-II adapter emulator v3.0.1 started on pseudo-tty port "/dev/pts/3".

Welcome to the ELM327 OBD-II adapter emulator.
ELM327-emulator is running on /dev/pts/3
Type help or ? to list commands.

CMD> 2021-07-09 21:25:59,439 - root - CRITICAL - Error while processing '':
string index out of range
Traceback (most recent call last):
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 848, in run
    request_header, request_data, resp = self.handle_request(
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 1802, in handle_request
    if cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
IndexError: string index out of range

2021-07-09 21:25:59,440 - root - CRITICAL - Error while processing '':
string index out of range
Traceback (most recent call last):
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 848, in run
    request_header, request_data, resp = self.handle_request(
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 1802, in handle_request
    if cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
IndexError: string index out of range

2021-07-09 21:25:59,442 - root - INFO - Unknown ELM command: 'ELM327V1.5', header=7E0
2021-07-09 21:25:59,443 - root - INFO - Unknown ELM command: 'ELM327V1.5', header=7E0
2021-07-09 21:25:59,443 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,443 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,444 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,444 - root - INFO - Unknown ELM command: 'ELM327V1.5', header=7E0
2021-07-09 21:25:59,444 - root - INFO - Unknown ELM command: 'ELM327V1.5', header=7E0
2021-07-09 21:25:59,445 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,445 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,445 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,445 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,445 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,446 - root - INFO - Unknown ELM command: 'ELM327V1.5', header=7E0
2021-07-09 21:25:59,446 - root - CRITICAL - Error while processing '?':
string index out of range
Traceback (most recent call last):
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 848, in run
    request_header, request_data, resp = self.handle_request(
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 1802, in handle_request
    if cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
IndexError: string index out of range

2021-07-09 21:25:59,446 - root - CRITICAL - Error while processing '?':
string index out of range
Traceback (most recent call last):
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 848, in run
    request_header, request_data, resp = self.handle_request(
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 1802, in handle_request
    if cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
IndexError: string index out of range

2021-07-09 21:25:59,446 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,446 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,447 - root - CRITICAL - Error while processing '?':
string index out of range
Traceback (most recent call last):
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 848, in run
    request_header, request_data, resp = self.handle_request(
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 1802, in handle_request
    if cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
IndexError: string index out of range

2021-07-09 21:25:59,447 - root - CRITICAL - Error while processing '?':
string index out of range
Traceback (most recent call last):
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 848, in run
    request_header, request_data, resp = self.handle_request(
  File "/home/jwalton/ELM327-emulator/elm/elm.py", line 1802, in handle_request
    if cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
IndexError: string index out of range

2021-07-09 21:25:59,447 - root - WARNING - Invalid request: '>'
2021-07-09 21:25:59,447 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,447 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,447 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,447 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,447 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,448 - root - WARNING - Invalid request: '>ATZ'
2021-07-09 21:25:59,448 - root - INFO - Unknown ELM command: 'ELM327V1.5', header=7E0
<skip many similar blocks of messages>

2021-07-09 21:26:01,542 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,542 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,542 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,543 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,544 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,544 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,587 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,588 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,638 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,688 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,688 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,738 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,739 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,788 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,839 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,889 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,939 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,940 - root - WARNING - Invalid request: '>'
2021-07-09 21:26:01,990 - root - WARNING - Invalid request: '>'
<skip many similar messages>
```

My apologies if I a doing something wrong here.

noloader, 2021-07-10:
I can avoid the crash with the following, but the AT command still results in problems at the emulator. The emulator does not return a response, and the emulator produces a ton of warnings. The client still hangs.

```diff
$ diff -u elm/elm.py.orig elm/elm.py
--- elm/elm.py
+++ elm/elm.py
@@ -1799,7 +1799,7 @@
         if self.delay > 0:
             time.sleep(self.delay)
 
-        if cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
+        if len(cmd) >=2 and cmd[1] == 'T' and org_cmd.upper()[1] != 'T':  # AT or ST shall be unspaced
             logging.error("Improper AT or ST command %s.", repr(org_cmd))
             return header, cmd, ""
 
```

Ircama, 2021-07-10:
This looks like strange behaviour, I'll analyze it early next week.

noloader, 2021-07-10:
Thanks @Ircama,

If you want an account on this machine, then please let me know. I am happy to provide it with admin privileges. I'll need your SSH public key. Email to _noloader_, _gmail_ account.

Otherwise the machine is Core i7-8700 running Ubuntu 20.04, x86_64, fully patched. You should be able to duplicate the OS and software environment in a VM.

noloader, 2021-07-10:
@Ircama,

One other question... Sorry to have to ask. I am not a Python person...

Should `master_fd` in `elm.py` include `os.O_NOCTTY`?

```
self.master_fd = os.open(self.device_port, os.O_RDWR)
```

noloader, 2021-07-10:
@Ircama,

I was able to connect to the emulator using GNU's `screen` program. I was also able to send an `ATZ` and receive the identification string with `screen`.

I've been down this road before... If I recall properly, `screen` opens the device with a lot of unexpected options ("unexpected" from the way a serial port is typically configured). In fact, my lab notes even say to look at GNU Screen's `InitTTY` function.

I attached my test program in case you want to kick it around for testing: [elm327.c.zip](https://github.com/Ircama/ELM327-emulator/files/6795958/elm327.c.zip).

-----

```c
/* The settings below are from GNU Screen program and its  */
/* InitTTY. InitTTY is screen's startup code for the TTY.  */
int term_config_gnu_screen(int fd, unsigned int speed)
{
    struct termios tty;
    if (tcgetattr(fd, &tty) != 0) {
        log_error("term_config: tcgetattr: %s\n", strerror(errno));
        return errno;
    }

    // Hack... Screen determines this at runtime, and sometimes it is 0.
    int ttyflag = 1;

#if defined(BRKINT)
    tty.c_iflag |= BRKINT;
#endif

#if defined(IGNPAR)
    tty.c_iflag |= IGNPAR;
#endif

#if defined(ISTRIP) && 0
    tty.c_iflag |= ISTRIP;
#endif

#if defined(IXON)
    tty.c_iflag |= IXON;
#endif

    if (!ttyflag) {
#if defined(ICRNL)
        tty.c_iflag |= ICRNL;
#endif

#if defined(ONLCR)
        tty.c_oflag |= ONLCR;
#endif

#if defined(TAB3)
        tty.c_oflag |= TAB3;
#endif

#if defined(OXTABS)
        tty.c_oflag |= OXTABS;
#endif

#if defined(PARENB) && 0
        tty.c_cflag |= PARENB;
#endif

#if defined(OPOST)
        tty.c_oflag |= OPOST;
#endif
    }

#if defined(CS8)
    tty.c_cflag |= CS8;
#endif

#if defined(CREAD)
    tty.c_cflag |= CREAD;
#endif

#if defined(CLOCAL)
    tty.c_cflag |= CLOCAL;
#endif

#if defined(ECHOCTL)
    tty.c_lflag |= ECHOCTL;
#endif

#if defined(ECHOKE)
    tty.c_lflag |= ECHOKE;
#endif

    if (!ttyflag) {
#if defined(ISIG)
        tty.c_lflag |= ISIG;
#endif

#if defined(ICANON)
        tty.c_lflag |= ICANON;
#endif

#if defined(ECHO)
        tty.c_lflag |= ECHO;
#endif

    }
#if defined(ECHOE)
    tty.c_lflag |= ECHOE;
#endif

#if defined(ECHOK)
    tty.c_lflag |= ECHOK;
#endif

#if defined(IEXTEN)
    tty.c_lflag |= IEXTEN;
#endif

#if defined(VINTR)
# if (VINTR < MAXCC)
    tty.c_cc[VINTR] = Ctrl('C');
# endif
#endif

#if defined(VQUIT)
# if (VQUIT < MAXCC)
    tty.c_cc[VQUIT] = Ctrl('\\');
# endif
#endif

#if defined(VERASE)
# if (VERASE < MAXCC)
    tty.c_cc[VERASE] = 0x7f;    /* DEL */
# endif
#endif

#if defined(VKILL)
# if (VKILL < MAXCC)
    tty.c_cc[VKILL] = Ctrl('U');
# endif
#endif

#if defined(VEOF)
# if (VEOF < MAXCC)
    tty.c_cc[VEOF] = Ctrl('D');
# endif
#endif

#if defined(VEOL)
# if (VEOL < MAXCC)
    tty.c_cc[VEOL] = VDISABLE;
# endif
#endif

#if defined(VEOL2)
# if (VEOL2 < MAXCC)
    tty.c_cc[VEOL2] = VDISABLE;
# endif
#endif

#if defined(VSWTCH)
# if (VSWTCH < MAXCC)
    tty.c_cc[VSWTCH] = VDISABLE;
# endif
#endif

#if defined(VSTART)
# if (VSTART < MAXCC)
    tty.c_cc[VSTART] = Ctrl('Q');
# endif
#endif

#if defined(VSTOP)
# if (VSTOP < MAXCC)
    tty.c_cc[VSTOP] = Ctrl('S');
# endif
#endif

#if defined(VSUSP)
# if (VSUSP < MAXCC)
    tty.c_cc[VSUSP] = Ctrl('Z');
# endif
#endif

#if defined(VDSUSP)
# if (VDSUSP < MAXCC)
    tty.c_cc[VDSUSP] = Ctrl('Y');
# endif
#endif

#if defined(VREPRINT)
# if (VREPRINT < MAXCC)
    tty.c_cc[VREPRINT] = Ctrl('R');
# endif
#endif

#if defined(VDISCARD)
# if (VDISCARD < MAXCC)
    tty.c_cc[VDISCARD] = Ctrl('O');
# endif
#endif

#if defined(VWERASE)
# if (VWERASE < MAXCC)
    tty.c_cc[VWERASE] = Ctrl('W');
# endif
#endif

#if defined(VLNEXT)
# if (VLNEXT < MAXCC)
    tty.c_cc[VLNEXT] = Ctrl('V');
# endif
#endif

#if defined(VSTATUS)
# if (VSTATUS < MAXCC)
    tty.c_cc[VSTATUS] = Ctrl('T');
# endif
#endif

    if (ttyflag) {
        tty.c_cc[VMIN] = 1;
        tty.c_cc[VTIME] = 0;
    }

#if defined(TIOCKSET)
    tty.m_jtchars.t_ascii = 'J';
    tty.m_jtchars.t_kanji = 'B';
    tty.m_knjmode = KM_ASCII | KM_SYSSJIS;
#endif

    cfsetospeed(&tty, term_get_baud(speed));
    cfsetispeed(&tty, term_get_baud(speed));

    if (tcsetattr(fd, TCSANOW, &tty) != 0) {
        log_error("term_config: tcsetattr: %s\n", strerror(errno));
        return errno;
    }

    if (tcflush(fd, TCIOFLUSH) != 0)
    {
        log_warn("term_config: tcflush: %s\n", strerror(errno));
    }

    return 0;
}
```

noloader, 2021-07-10:
@Ircama,

I kind of got to the bottom of things... In my test program I needed to use a raw terminal, and not the terminal attributes taken from `/dev/pts/3`.

So I changed this:

```c
struct termios tty;
if (tcgetattr(fd, &tty) != 0) {
    printf("Failed to get terminal attributes: %s\n", strerror(errno));
    goto cleanup_and_exit;
}
```

to this:

```c
struct termios tty;
cfmakeraw(&tty);
```

and things worked as expected.

Sorry about the extra noise.

Ircama, 2021-07-14:
Please recheck this issue after `python3 -m pip install --upgrade git+https://github.com/ircama/ELM327-emulator`

noloader, 2021-07-14:
Thanks @Ircama ,

> Please recheck this issue after python3 -m pip install --upgrade git+https://github.com/ircama/ELM327-emulator

I'm using master, which looks like it has the changes.

Using `cfmakeraw` to configure a terminal works as expected. It worked before and after the commits.

Using `tcgetattr` to configure a terminal no longer crashes/throws exception after the commits. However, the stream of messages is still present in the emulator, and the client still hangs on `ATZ`.

Ircama, 2021-08-13:
Thanks @noloader for these notes and sorry for this this long wait before my feedback.

I have created a new branch named [slave_raw](https://github.com/Ircama/ELM327-emulator/tree/slave_raw) with a possible fix that I need to test with all the applications that I have already used for the previous verifications.

Please recheck your issue after `python3 -m pip install --upgrade git+https://github.com/ircama/ELM327-emulator.git@slave_raw`.

Regards

closed 2021-09-28 by Ircama