Ircama/ELM327-emulator #19: running elm327 as a CAN interface

stephankaufmann, 2021-09-13:
Hi Ircama, i love your project! Is it possible to run your sim with a can interface. We have a linux hw with CAN/K-Line and would love to support elm327 commands. Stephan

Ircama, 2021-09-13:
Possibly in the future. K-Line is not a huge work, but I need to find enough time.

stephankaufmann, 2021-09-15:
Thank you for the quick reply! We would just be happy with CAN in the beginning. Can we have a chat about the effort from your side? stephan@ridelink.com

grayvader, 2023-10-01:
I second the request!

Ircama, 2026-09-17:
The emulator now exposes the two buses: -c/--slcan PORT emulates a CAN interface and -K/--kline PORT emulates an ISO 9141-2/ISO 14230 K-Line ECU. What is still missing is the opposite direction.

closed 2026-09-17 by Ircama