Ircama/ELM327-emulator #20: Inject additional PIDs with data from external sensors to the existing PIDs from the vehicle OBD adapter

lademeister, 2021-11-22:
I understand that your (great!) software can do that, but I would need something to get started.
What I want to do:
add external (physical) sensors to a Raspberry Pi (like fuel pressure and boost pressure, exhaust temperature etc) and add add PIDs for these sensors that are not existing in the ECU(s) that the vehicle has.
I have lets say an engine ECU that provides RPM, coolant temp etc. and I have an OBD2-interface that reads that data.
I can loop-through the data from OBD interface and manipulate the data - I understood and tested that part.
Actually, for ranch testing I started your car scenario, and looped that through and modified the PID values, as I don't have the actual car at my desk.
Works so far - but I don't get what I need to do to modify PID values from lets say text files containing actual measurement data from external sensors that I added. I only understand how do do command-line based modifications to the live values.
It would be great to get a short example to start with... :-)

lademeister, 2021-12-01:
How can I set values to existing pid with a shellscript or python script? 
(not via CMD prompt)?

Ircama, 2022-01-15:
Sorry for being late in the answer; my free time is very limited.

> put the modified interface to /dev/pts/1

How you do this might not be clear. Notice anyway that if you are using ELM327-emulator as a proxy through its [forwarder options]( https://github.com/Ircama/ELM327-emulator#forwarder-options), I would not suggest this because that method has been just developed in order to verify and improve the dictionary during the customization phase and not for standard on-line filtering operation.

> I don't get what I need to do to modify PID values (or add PIDs) with data from lets say text files containing actual measurement data from external sensors that I added.

I understand that you want to dynamically change the values which are statically predefined in [obd_message.py]( https://github.com/Ircama/ELM327-emulator/blob/master/elm/obd_message.py).

I see at least four possibilities.

### MODE 1 – using obd_message.py

This method adds a function to *obd_message.py* to dynamically load a configuration file.

Example.

Create a configuration file (e.g., YAML format):

```bash
$ echo -e "---\nINTAKE_PRESSURE: '76'" > /tmp/pid.yaml
```

Edit *obd_message.py* and add the following at the top of the file:

```python
import yaml

def load_pid(pid):
    with open('/tmp/pid.yaml', 'rb') as f:
        conf = yaml.load(f.read(), Loader=yaml.BaseLoader)
        return conf[pid]
```

Take for instance the INTAKE_PRESSURE pid, which returns the following by default.

```
CMD> test 010B
______Raw command:_______________
'<pos_answer>73</pos_answer>'

______Command output:____________
'41 0B 73 \r\r>'
CMD>
```

Edit the INTAKE_PRESSURE pid in *obd_message.py* with the following:

```python
        'INTAKE_PRESSURE': {
            'Request': '^010B' + ELM_FOOTER,
            'Descr': 'Intake Manifold Pressure',
            'ResponseFooter': lambda self, cmd, pid, uc_val: PA(load_pid(pid))
        }, 
```

Now the 'INTAKE_PRESSURE' returns the value defined in */tmp/pid.yaml* by reading the file content each time the pid is requested.

### MODE 2 – using batch mode

This method loads predefined parameters via [batch mode](https://github.com/Ircama/ELM327-emulator#elm327-emulator-batch-mode).

Populate a bash configuration file:

```bash
echo "INTAKE_PRESSURE='76'" > /tmp/pid.sh
```

Develop a batch program reading that file periodically (trivial 1-line example):

```bash
while true;do . /tmp/pid.sh; echo "emulator.answer['INTAKE_PRESSURE']='<pos_answer>$INTAKE_PRESSURE</pos_answer>'";sleep 1;done | elm -b /tmp/elm.out
```

Each second the configuration file is read and the INTAKE_PRESSURE answer is dynamically updated.

### MODE 3 – using the python interactive mode

This method performs the interactive loading of parameters via [Python Interactive mode]( https://github.com/Ircama/ELM327-emulator#interactive-mode).

Create a configuration file (e.g., YAML format):

```bash
$ echo -e "---\nINTAKE_PRESSURE: '76'" > /tmp/pid.yaml
```

Create the following Python program:

```python
from elm import Elm
import time

import yaml

def load_pid(pid):
    with open('/tmp/pid.yaml', 'rb') as f:
        conf = yaml.load(f.read(), Loader=yaml.BaseLoader)
        return conf[pid]

with Elm() as session:
    # interactive monitoring
    pty = session.get_pty()
    print(f"Used port: {pty}")
    while True:
        session.answer['INTAKE_PRESSURE']=(
            '<pos_answer>' +
            load_pid('INTAKE_PRESSURE') +
            '</pos_answer>')
        time.sleep(0.5) # example
```

This program updates the INTAKE_PRESSURE *answer* evey second by reading its value from the YAML configuration file.

### MODE 4 – developing a task

Create a configuration file (e.g., YAML format):
```bash
$ echo -e "---\nINTAKE_PRESSURE: '76'" > /tmp/pid.yaml
```

Edit the INTAKE_PRESSURE pid in *obd_message.py* by invoking a task named *task_intake_pressure* with the following:

```python
        'INTAKE_PRESSURE': {
            'Request': '^010B' + ELM_FOOTER,
            'Descr': 'Intake Manifold Pressure',
            'Task': 'task_intake_pressure'
        }, 
```

Create a file named "task_intake_pressure.py" in the *plugin* directory with the following content:

```python
from elm import Tasks
import yaml

def load_pid(pid):
    with open('/tmp/pid.yaml', 'rb') as f:
        conf = yaml.load(f.read(), Loader=yaml.BaseLoader)
        return conf[pid]

class Task(Tasks):
    def run(self, cmd, *_):
        return self.PA(load_pid(self.pid)), Tasks.RETURN.TERMINATE, None
```

Execution example:

```
$ elm
2022-01-15 12:45:09,259 - root - INFO -
ELM327 OBD-II adapter emulator v3.0.2 started on pseudo-tty port "/dev/pts/1".
Welcome to the ELM327 OBD-II adapter emulator.
ELM327-emulator is running on /dev/pts/1
Type help or ? to list commands.

CMD> test 010B
______Raw command:_______________
'<pos_answer>76</pos_answer>'

______Command output:____________
'41 0B 76 \r\r>'

CMD> os.system("{ echo '---'; echo \"INTAKE_PRESSURE: '74'\";} > /tmp/pid.yaml")
0

CMD> test 010B
______Raw command:_______________
'<pos_answer>74</pos_answer>'

______Command output:____________
'41 0B 74 \r\r>'

CMD>
```
By far the task method is the most flexible one.

lademeister, 2022-01-15:
By far this is the best answer anyone could get. Maybe the best on the internet.

Thank you so much for your time! I will try that.

closed 2022-01-15 by lademeister