Ircama/ELM327-emulator #21: briefly start and crash after "supposedly" changing the baud rate in GUI

serpico, 2021-11-30:
Xubuntu 21.04

I might have changed the baud rate on the GUI at one point and now when I launch by typing  

`scantool`

I get 

```arno@arno-ThinkPad-X220:~$ scantool
Tue Nov 30 04:51:13 2021
Version: 2.1 for POSIX

Initializing All Modules...
---------------------------
Initializing Allegro... OK

Installing Timers... OK
Installing Keyboard... OK
Installing Mouse... OK
Loading Preferences... OK
Trying Windowed Graphics Mode... OK
Loading Data File... OK
Initializing Serial Module... OK
Opening COM4... OK

Displaying Main Menu...
-----------------------
write tty: Erreur d'entrée/sortie
*** buffer overflow detected ***: terminated
Shutting down Allegro due to signal #6
Abandon (core dumped)
arno@arno-ThinkPad-X220:~$ 
```
I can see the GUI window appear for an instant then disappear, so I can't reverse the setting.

I tried to look for a config file were the values for baud rate and COM port would be written/stored, to no avail.

I'd appreciate if you could point towards something.

Thanks

serpico, 2021-12-03:
wrong repo

closed 2021-12-03 by serpico

tit6, 2024-02-27:
merci pour tout j'ai enfin compris merci