Ircama/ELM327-emulator #23: Unknown request: '0160', header=7E0

steigerth, 2022-01-02:
I want to use this ELM327 emulator in connection with [Perl OBD-II Logger](https://sourceforge.net/projects/pobd-logger/).
Unfortunately right from the beginning I encounter the following error: " Unknown request: '0160', header=7E0"

```text
CMD> 2022-01-02 09:29:01,777 - root - DEBUG - Received 'ATZ'
2022-01-02 09:29:01,777 - root - DEBUG - Handling: 'ATZ', header '7E0', ECU '7E0'
2022-01-02 09:29:01,777 - root - DEBUG - Instantiating ECU task "Default ECU Task module" for ECU "7E0"
2022-01-02 09:29:01,780 - root - DEBUG - Description: AT RESET, PID AT_RESET (ATZ)
2022-01-02 09:29:01,780 - root - DEBUG - Resetting counters and sleeping for 0.5 seconds
2022-01-02 09:29:02,300 - root - DEBUG - Reset and sleep 0.5 seconds
2022-01-02 09:29:02,300 - root - DEBUG - Processing: '<writeln></writeln><writeln></writeln><writeln>ELM327 v1.5</writeln>'
2022-01-02 09:29:02,300 - root - DEBUG - Write: '\r\rELM327 v1.5\r\r>'
2022-01-02 09:29:02,558 - root - DEBUG - Received 'ATE0'
2022-01-02 09:29:02,559 - root - DEBUG - Handling: 'ATE0', header '7E0', ECU '7E0'
2022-01-02 09:29:02,559 - root - DEBUG - Description: AT ECHO, PID AT_ECHO (ATE0)
2022-01-02 09:29:02,559 - root - DEBUG - set ECHO ON/OFF : False
2022-01-02 09:29:02,559 - root - DEBUG - Processing: '<writeln>OK</writeln>'
2022-01-02 09:29:02,559 - root - DEBUG - Write: 'OK\r\r>'
2022-01-02 09:29:02,820 - root - DEBUG - Received 'ATL0'
2022-01-02 09:29:02,823 - root - DEBUG - Handling: 'ATL0', header '7E0', ECU '7E0'
2022-01-02 09:29:02,823 - root - DEBUG - Description: AT LINEFEEDS, PID AT_LINEFEEDS (ATL0)
2022-01-02 09:29:02,823 - root - DEBUG - set LINEFEEDS ON/OFF : False
2022-01-02 09:29:02,824 - root - DEBUG - Processing: '<writeln>OK</writeln>'
2022-01-02 09:29:02,824 - root - DEBUG - Write: 'OK\r\r>'
2022-01-02 09:29:03,078 - root - DEBUG - Received 'ATS0'
2022-01-02 09:29:03,078 - root - DEBUG - Handling: 'ATS0', header '7E0', ECU '7E0'
2022-01-02 09:29:03,079 - root - DEBUG - Description: AT Spaces off or on, PID AT_SPACES (ATS0)
2022-01-02 09:29:03,079 - root - DEBUG - set SPACES False
2022-01-02 09:29:03,079 - root - DEBUG - Processing: '<writeln>OK</writeln>'
2022-01-02 09:29:03,079 - root - DEBUG - Write: 'OK\r\r>'
2022-01-02 09:29:03,336 - root - DEBUG - Received 'ATI'
2022-01-02 09:29:03,336 - root - DEBUG - Handling: 'ATI', header '7E0', ECU '7E0'
2022-01-02 09:29:03,337 - root - DEBUG - Description: AT ELM327 version string, PID AT_I (ATI)
2022-01-02 09:29:03,337 - root - DEBUG - Processing: '<writeln>ELM327 v1.5</writeln>'
2022-01-02 09:29:03,337 - root - DEBUG - Write: 'ELM327 v1.5\r\r>'
2022-01-02 09:29:03,337 - root - DEBUG - Received '0100'
2022-01-02 09:29:03,337 - root - DEBUG - Handling: '0100', header '7E0', ECU '7E0'
2022-01-02 09:29:03,337 - root - DEBUG - Description: PIDS_A, PID ELM_PIDS_A (0100)
2022-01-02 09:29:03,338 - root - DEBUG - Processing: '<string>SEARCHING...</string><exec>time.sleep(3)</exec><writeln></writeln><header>7EA</header><size>06</size><data>41 00 98 3A 80 13</data><pos_answer>BE 3F A8 13</pos_answer>'
2022-01-02 09:29:03,338 - root - DEBUG - Write: 'SEARCHING...'
2022-01-02 09:29:06,352 - root - DEBUG - Evaluated command: time.sleep(3) -> None
2022-01-02 09:29:06,352 - root - DEBUG - Write: '\r4100983A8013\r4100BE3FA813\r\r>'
2022-01-02 09:29:09,190 - root - DEBUG - Received '0100'
2022-01-02 09:29:09,190 - root - DEBUG - Handling: '0100', header '7E0', ECU '7E0'
2022-01-02 09:29:09,190 - root - DEBUG - UDS P3 timer expired: running stop() method for ECU task "Default ECU Task module", ECU="7E0".
2022-01-02 09:29:09,190 - root - DEBUG - Description: PIDS_A, PID ELM_PIDS_A (0100)
2022-01-02 09:29:09,191 - root - DEBUG - Processing: '<header>7EA</header><size>06</size><data>41 00 98 3A 80 13</data><pos_answer>BE 3F A8 13</pos_answer>'
2022-01-02 09:29:09,191 - root - DEBUG - Write: '4100983A8013\r4100BE3FA813\r\r>'
2022-01-02 09:29:09,191 - root - DEBUG - Received '0120'
2022-01-02 09:29:09,191 - root - DEBUG - Handling: '0120', header '7E0', ECU '7E0'
2022-01-02 09:29:09,191 - root - DEBUG - Description: PIDS_B, PID ELM_PIDS_B (0120)
2022-01-02 09:29:09,191 - root - DEBUG - Processing: '<header>7EA</header><size>06</size><data>41 20 80 01 A0 01</data><pos_answer>90 15 B0 15</pos_answer>'
2022-01-02 09:29:09,191 - root - DEBUG - Write: '41208001A001\r41209015B015\r\r>'
2022-01-02 09:29:09,192 - root - DEBUG - Received '0140'
2022-01-02 09:29:09,192 - root - DEBUG - Handling: '0140', header '7E0', ECU '7E0'
2022-01-02 09:29:09,192 - root - DEBUG - Description: PIDS_C, PID ELM_PIDS_C (0140)
2022-01-02 09:29:09,192 - root - DEBUG - Processing: '<header>7EA</header><size>06</size><data>41 40 44 CC 00 21</data><pos_answer>7A 1C 80 00</pos_answer>'
2022-01-02 09:29:09,192 - root - DEBUG - Write: '414044CC0021\r41407A1C8000\r\r>'
2022-01-02 09:29:09,192 - root - DEBUG - Received '0160'
2022-01-02 09:29:09,192 - root - DEBUG - Handling: '0160', header '7E0', ECU '7E0'
2022-01-02 09:29:09,193 - root - INFO - Unknown request: '0160', header=7E0
2022-01-02 09:29:09,193 - root - DEBUG - Processing: '<writeln>NO DATA</writeln>'
2022-01-02 09:29:09,193 - root - DEBUG - Write: 'NO DATA\r\r>
```
`

Ircama, 2022-01-04:
The default scenario has a very limited configuration. You should add a "0160" pid similar to the ["0140" one](https://github.com/Ircama/ELM327-emulator/blob/44978063006dd360dca9fe6ae6e0d2736241c786/elm/obd_message.py#L1575-L1584), or, more simply, switch to the ["car" scenario](https://github.com/Ircama/ELM327-emulator/blob/44978063006dd360dca9fe6ae6e0d2736241c786/README.md?plain=1#L79) which simply [sets it to no data](https://github.com/Ircama/ELM327-emulator/blob/44978063006dd360dca9fe6ae6e0d2736241c786/elm/obd_message.py#L1770-L1774).

steigerth, 2022-01-04:
Thank for the response. The perl obd logger stopes also if I switch to the car scenario. But I found a solution. I skipped the request for 0160 in the perl obd logger. With that the simulation works. Thank you very much for this fantastic simulator. It looks like it is highly customizable.

closed 2022-01-04 by steigerth