Ircama/ELM327-emulator #24: "NO DATA" for PIDs that are clearly in the obd_message.py file

aomanchuria, 2022-01-08:
as stated, I am trying to get MODE 21 for specifically a Prius C and I see the PIDs listed, but I don't understand how to get them.

```
CUSTOM_V01': {
            'Request': '^2181' + ELM_FOOTER,
            'Descr': 'Battery Block Voltage -V01',
            'Equation': '(A * 256 + B) * 79.99 / 65535',
            'Min': '12',
            'Max': '20',
            'Unit': 'V',
            'Header': ECU_ADDR_H,
            'Response': [
                        HD(ECU_R_ADDR_H) + SZ('10') + DT('22 61 81 37 7C 37 1A') +
                        HD(ECU_R_ADDR_H) + SZ('21') + DT('38 08 38 08 38 BC 38') +
                        HD(ECU_R_ADDR_H) + SZ('22') + DT('BC 37 95 37 95 37 33') +
                        HD(ECU_R_ADDR_H) + SZ('23') + DT('37 22 38 49 38 5A 37') +
                        HD(ECU_R_ADDR_H) + SZ('24') + DT('53 37 95 AF 3B 09 74'),
                        HD(ECU_R_ADDR_H) + SZ('10') + DT('22 61 81 38 72 38 20') +
```

I setup a port pair between COM3 and COM4 using comOcom. the I connected your emulator to COM3 and connected YAT to COM4. I then typed for example 'ATVR' and I get a correctly faked voltage reading. However if I type 2181 I get "No Data"

`2101
2101<CR>7E8 10 1B 61 01 6F 00 44 02 <CR>7E8 21 C6 28 38 61 4B 17 21 <CR>7E8 22 31 00 2B 01 30 83 31 <CR>7E8 23 04 00 31 00 5B 39 5F <CR><CR>><LF>
2161
2161<CR>NO DATA<CR><CR>><LF>
2130
2130<CR>NO DATA<CR><CR>><LF>
2129
2129<CR>NO DATA<CR><CR>><LF>
2161
2161<CR>NO DATA<CR><CR>><LF>
2181
2181<CR>NO DATA<CR><CR>><LF>
ATRV
ATRV<CR>13.1V<CR><CR>><LF>`

`<html>
<head>
	
	
Another thing I don't understand is how to request for example 2103 if there are two such PIDS, how does one use the header?

	<title></title>
	<meta name="generator" content="LibreOffice 7.0.2.2 (Windows)"/>
	<style type="text/css">
		body,div,table,thead,tbody,tfoot,tr,th,td,p { font-family:"Liberation Sans"; font-size:x-small }
		a.comment-indicator:hover + comment { background:#ffd; position:absolute; display:block; border:1px solid black; padding:0.5em;  } 
		a.comment-indicator { background:red; display:inline-block; border:1px solid black; width:0.5em; height:0.5em;  } 
		comment { display:none;  } 
	</style>
	
</head>

<body>

Name | ShortName | ModeAndPID | Equation | Min Value | Max Value | Units | Header
-- | -- | -- | -- | -- | -- | -- | --
FR Wheel Speed | FR WS | 2103 | A * 32 / 25 | 0 | 200 | km/h | 7B0
FL Wheel Speed | FL WS | 2103 | B * 32 / 25 | 0 | 200 | km/h | 7B0
RR Wheel Speed | RR WS | 2103 | C * 32 / 25 | 0 | 200 | km/h | 7B0
RL Wheel Speed | RL WS | 2103 | D * 32 / 25 | 0 | 200 | km/h | 7B0
Fuel System Status #1 (OL=1,CL=2,OLDrive=4,OLFault=8,CLFault=16) | FSS1 | 2103 | A | 1 | 16 |   | 7
Short FT #1 | Short FT1 | 2103 | C * 199.2 / 255 - 100 | -100 | 99.2 | % | 7
Long FT #1 | Long FT1 | 2103 | D * 199.2 / 255 - 100 | -100 | 99.2 | % | 7
IGN Advance | IGN | 2103 | E / 2 - 64 | -64 | 63.5 | deg | 7

</body>

</html>
`

Ircama, 2022-01-09:
You need to select `scenario car` and change the header to *ECU_ADDR_H = "7E2"*:

```
$ elm
2022-01-09 16:47:30,777 - root - INFO -

ELM327 OBD-II adapter emulator v3.0.2 started on pseudo-tty port "/dev/pts/1".

Welcome to the ELM327 OBD-II adapter emulator.
ELM327-emulator is running on /dev/pts/1
Type help or ? to list commands.

CMD> scenario car
Emulator scenario switched to 'car'
CMD> test atsh7e2
______Raw command:_______________
'<writeln>OK</writeln>'

______Command output:____________
'OK\r\r>'
CMD> test 2181
______Raw command:_______________
'<header>7EA</header><size>10</size><data>22 61 81 37 7C 37 1A</data><header>7EA</header><size>21</size><data>38 08 38 08 38 BC 38</data><header>7EA</header><size>22</size><data>BC 37 95 37 95 37 33</data><header>7EA</header><size>23</size><data>37 22 38 49 38 5A 37</data><header>7EA</header><size>24</size><data>53 37 95 AF 3B 09 74</data>'

______Command output:____________
'22 61 81 37 7C 37 1A \r38 08 38 08 38 BC 38 \rBC 37 95 37 95 37 33 \r37 22 38 49 38 5A 37 \r53 37 95 AF 3B 09 74 \r\r>'
CMD>
```

aomanchuria, 2022-01-09:
Thanks man! I didn't know how that was done. This worked.

closed 2022-01-10 by Ircama