Ircama/ELM327-emulator #25: any workarounds for getting it run with virtual COM Ports in Windows?

rs38, 2022-01-20:
com0com and probably any other driver is not able to work without proper driver signature and that is true for nearly every windows in 2022 (Win10+ and 64Bit).

I don't think that disabling driver signature enforcement is a good idea, if even possible.

At least I guess it should be mentioned that it won't work.

seel also #3

Ircama, 2022-01-21:
I am not aware of a Windows capability similar to the Linux pseudo-terminal function used by *ELM327-emulator*.

Anyway, implementing null-modem emulation on Windows is beyond the intent of *ELM327-emulator*. A number of software tools already exist to perform this. And when using TCP/IP or Bluetooth, the null-modem feature is not even required.

As an option, I think that hardware null-modem USB cables might be available.

A hardware adapter (to be connected to two physical USB ports) can be easily built through two inexpensive USB-to-TTL serial UART modules, by cross-connecting them via their TX and RX lines (provided to choose hw adapters offering signed Windows drivers).

rs38, 2022-01-21:
I did not even think about the existence of null-modem USB cables, but nice idea. Eventually I got it the emulator to run , so from a community stand point it would only make sense to make it better and create a PR, right? :)

Ircama, 2022-01-21:
Yes, sure. If you think a PR on the documentation could be useful, no problem.

Perhaps the cheapest option is to directly build the adapter (very few dollars/Euros):
- USB-to-serial RS232 cable - null modem - serial RS232 cable to USB cable
or
- USB cable - FTDI USB to TTL module - cross connection - FTDI USB to TTL module - USB cable

syedtaqi95, 2022-10-15:
@rs38 can you please share how you got the emulator to work on Windows? I've just installed the package and ran into the same issue as you.