Ircama/ELM327-emulator #29: How to emulate Bluetooth on MacOS?

lygialiem, 2022-12-22:
Hi team, thank you for your wonderful library.

About Bluetooth connection methods, I only saw Bluetooth with Window and Linux/Unix. What about MacOS? Specifically, I'm developing iOS app that need an emulator to test connect with Bluetooth. Can this library do that? Hope to see your response soon.

Thank you and Best Regard.

lygialiem, 2022-12-27:
It's seem like in higher MacOS: Ventura, Apple doesn't allow to setup Bluetooth serial tty port. There is no options or settings in Bluetooth Settings. I have browsed on website whole day and there is no solutions for that. It's hard but can you take a quick research on that so that we can jump to a conclusion.

I also setup Bluetooth in Window PC and my iOS iPhone still can't connect to Window PC (unsupported).

Thank you and Best Regard.

Ircama, 2026-09-17:
With version 4.0.0, Bluetooth SPP emulation is now built in.

closed 2026-09-17 by Ircama