Ircama/ELM327-emulator #30: Connect Physical device

Peskedor, 2022-12-24:
Thanks for this awesome library!

I do understand that it's rather meant to test your own software, however I'm trying to feed a device in my car with the OBD Data from this script. The device has a regular OBD plug and reads data from my car while driving and takes actions based on the status of the car. I'm searching for 2 days now for a possibility to connect the OBD plug somehow with my Raspberry Pi / any Linux computer and use this script to emulate my car. Do you have any idea/experience how this could be archived? I would need to somehow connect the OBD plug with my Linux device and feed it with the data of your emulator.

Ircama, 2022-12-27:
*ELM327-emulator* does not implement the physical link layer, which depends on the hardware (e.g., K-Line, CAN bus).

kenreider, 2024-04-17:
> Thanks for this awesome library!
>
> I do understand that it's rather meant to test your own software, however I'm trying to feed a device in my car with the OBD Data from this script. The device has a [...]

Did you ever figure this out? I'm trying to do what you did too, but have the same issue/confusion. Thanks!