Ircama/ELM327-emulator #31: How to simulate Diagnostic Trouble Code (Mode/Services 03, 07, 0A)?

lygialiem, 2023-02-20:
Hi team.

I just wonder if this repo can simulate DTC?

Thank you and best regard.

nagibbinazad, 2023-09-07:
Any solution to this ⬆️?

gabrielgrijincu, 2024-02-04:
This would be awesome. Everything I get for querying `0x03` is `4300` (no codes).

Ircama, 2024-02-24:
Anything that is not directly provided by current software can be developed and then added; I accept PR.

travisryan, 2024-07-27:
I too would like to see this. I'm not sure what the scenario, etc is for but seems like there is some sort of ability to extend this.

Mirecos, 2025-04-28:
Hey, a year later i'm coming up on this issue. I've been trying to find a way to manually add diagnostic code to the emulator responses. 

I've manually added them to the 'GET_DTC' command answer, in the file [obd_message.py](https://github.com/Ircama/ELM327-emulator/blob/master/elm/obd_message.py) file. So, if you only want to test retrieving DTC's with the 'GET_DTC' command, you can give it a shot.

You can replace the Response from this :
`'Response': HD(ECU_R_ADDR_E) + SZ('02') + DT('43 00')`
To :
`'Response': HD(ECU_R_ADDR_E) + SZ('05') + DT('43 03 00 02 00'), # returns a P3302 DTC for example`

Note that it only allows you to test the 'GET_DTC' command. Generating a DTC in a real situation probably leads to multiple more complex logics.

I'm still new to OBD systems, any tips / docs is appreciated.
I will try to give more informations if I manage to create a more viable solutions.