Ircama/ELM327-emulator #34: legacy protocol support?

csurf, 2023-06-20:
I tried running the emulator & setting it to protocol # '3' (ISO-9141-2), but it simply ignored the 'AT SP3' command and kept using CAN. So, it apparently can only speak CAN.

Is there a way to configure it for other 'legacy' protocols? Do all of the communications & messaging need to be created from scratch to support such protocols?

Ircama, 2023-06-29:
ELM327-emulator supports a subset of the ISO 15765-4 protocol (CAN 11/500).

Currently, ATSP just sets the internal `cmd_proto` counter, which can be queried with the `counters` command, but it is not used.

The code offers the possibility of future extensions, like the support of K-line.

theFus, 2023-12-10:
Hello,
what  would it mean in terms of work to implement k-line iso 9141 protocol? I have basic knowledge in python and arduino c++ code, and i'm learning about k-line. I can imagine to work for it, but i have doubts considering my competences. On the other hand i can learn fast.
cheers

Ircama, 2026-09-17:
K-Line/ISO 14230 is now available as an emulated interface in v4.0.0, not only as a scenario: python -m elm -s car -K COM15 (or -K /dev/pts/N) makes the emulator drive the K-Line directly. ATSP3/4/5 still does not switch the framing of the CAN-oriented entries. A protocol layer driven by ATSP/ATTP remains a structural change.

closed 2026-09-17 by Ircama