Ircama/ELM327-emulator #40: How to emulate multi frame response ?

MikkM, 2024-03-24:
Hi, how can i add MODE 01 PID 78 (EGT sensors) to emulator that gives back multi frame response?

Example from my car:
>0178
7E8100B41780D061F01
7E8219003DA02A60000

Ircama, 2024-03-25:
Hi, I don't remember exactly, but it should be possible by editing obd_message.py.

Ircama, 2026-09-17:
```
ObdMessage = {
    'egt': {
        'EXHAUST_GAS_TEMPERATURE': {
            'Request': '^0178' + ELM_FOOTER,
            'Descr': 'Exhaust Gas Temperature (multi-frame example)',
            # 11 payload bytes: 41 78 (positive answer of mode 01 PID 78)
            # followed by 9 data bytes, as in the trace of the issue
            'ResponseFooter': lambda self, cmd, pid, uc_val: iso_tp_frames(['41', '78', '0D', '06', '1F', '01', '90', '03', 'DA', '02', 'A6'])
        }
    }
}
```

closed 2026-09-17 by Ircama