Ircama/ELM327-emulator #43: COM3 Problem

donovan-maker, 2024-07-14:
When I try to run the program:

2024-07-14 18:12:06,015 - root - CRITICAL - Error while opening serial COM 'COM3':
could not open port 'COM3': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

Help?

bullkanox, 2024-10-07:
1- Create port with com0com, example "COM10"
2- Run emulator with this command:             elm -p COM10

Mohid-Anwar, 2024-11-27:
The issue here is that you are not using a Null modem Emulator 
if you're using windows 10 or 11 Please use the Following link to access the installer http://radio.jsh.de/com0com.zip
Make sure to tick the 1st and 3rd checkbox while installing 
Then try running the program

kahlkevin, 2025-07-22:
PSA: com0com isn't going to work on Windows 11 until you disable Secure Boot on your system. This is because the (kernel mode) drivers are not signed by Microsoft's WHQL, and will therefore be rejected by the kernel with Error 52. Sure, you can _install_ them just fine if you agree to trust the code signing certificate... you just won't be able to actually _use_ them at run time until Secure Boot (and therefore driver verification enforcement) is disabled.

After you disable Secure Boot, com0com works just fine on Win11. However, I do recommend downloading from the proect home at SourceForge, and not some random other location such as radio.jsh.de.